esp-rs/esp-generate #19: Code which is generated does not compile

Krensi, 2024-10-23:
Hi! 

Running following command: `esp-generate --chip esp32c3 -o ble -o wifi -o embassy -o alloc -o probe-rs -o wokwi -o ci xiao-no-std-esp-gen` leads to a project which does not compile. 

Compiler output:
```
WARNING: use --release
  We *strongly* recommend using release profile when building esp-hal.
  The dev profile can potentially be one or more orders of magnitude
  slower than release, and may cause issues with timing-senstive
  peripherals and/or devices.

   Compiling xiao-no-std-esp-gen v0.1.0 (D:\Software\esp32_rust\xiao-no-std-esp-gen)
error[E0277]: the trait bound `EspWifiInitFor: EspWifiTimerSource` is not satisfied
   --> src\bin\async_main.rs:31:9
    |
29  |     let _init = esp_wifi::init(
    |                 -------------- required by a bound introduced by this call
30  |         esp_wifi::EspWifiInitFor::Wifi,
31  |         esp_wifi::EspWifiInitFor::Ble,
    |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `esp_wifi::IntoAnyTimer` is not implemented for `EspWifiInitFor`, which is required by `EspWifiInitFor: EspWifiTimerSource`
    |
    = help: the following other types implement trait `esp_wifi::IntoAnyTimer`:
              Alarm<'_, T, DM, COMP, UNIT>
              AnyTimer
              esp_hal::timer::timg::Timer<T, DM>
    = note: required for `EspWifiInitFor` to implement `EspWifiTimerSource`
note: required by a bound in `esp_wifi::init`
   --> C:\Users\ckren\.cargo\registry\src\index.crates.io-6f17d22bba15001f\esp-wifi-0.10.1\src/lib.rs:406:17
    |
404 | pub fn init(
    |        ---- required by a bound in this function
405 |     init_for: EspWifiInitFor,
406 |     timer: impl EspWifiTimerSource,
    |                 ^^^^^^^^^^^^^^^^^^ required by this bound in `init`

error[E0061]: this function takes 4 arguments but 5 arguments were supplied
   --> src\bin\async_main.rs:29:17
    |
29  |     let _init = esp_wifi::init(
    |                 ^^^^^^^^^^^^^^
...
32  |         timg1.timer0,
    |         ------------ unexpected argument #3 of type `esp_hal::timer::timg::Timer<TimerX<TIMG1>, Blocking>`
    |
note: function defined here
   --> C:\Users\ckren\.cargo\registry\src\index.crates.io-6f17d22bba15001f\esp-wifi-0.10.1\src/lib.rs:404:8
    |
404 | pub fn init(
    |        ^^^^
help: remove the extra argument
    |
31  -         esp_wifi::EspWifiInitFor::Ble,
32  -         timg1.timer0,
31  +         esp_wifi::EspWifiInitFor::Ble,
    |
```

I also mentioned this in the matrix chat:
https://matrix.to/#/!LdaNPfUfvefOLewEIM:matrix.org/$Cqev3HK8Nc5K5MYE3pwcDjRd_ijKB3PmmWr3Ueyqk6Q?via=matrix.org&via=tchncs.de&via=envs.net

bjoernQ, 2024-10-23:
Thanks for testing.

The problem is most probably caused by `-o ble -o wifi` - currently `ble` and `wifi` are mutual exclusive features (we might allow that in future). The TUI won't let you choose both but apparently, we don't correctly check the options given via CLI yet

Krensi, 2024-10-23:
Ah okay, thanks :-)

PR #23 (2024-10-25): Make the 'ble' and 'wifi' options mutually exclusive
Closes #19
closed 2024-10-25 by jessebraham