esp-rs/esp-generate #24: ERROR probe_rs::util::rtt: Error reading from RTT: The control block has been corrupted

Krensi, 2024-10-25:
Hi! Sorry for being annoying again. 😀 
Thanks for your work on the other tickets! 

I retried running `esp-generate --chip esp32c3 -o ble -o embassy -o alloc -o probe-rs -o wokwi -o ci xiao-no-std-esp-gen` with the [latest main](https://github.com/esp-rs/esp-generate/commit/ff981be8bbc2ba72eb864f2c3c267562c4a64dd9).
Now I get some output via RTT, however it reports an error soon after sending "Hello world!".

```
D:\Software\fillament_spool_lamp> cargo r --release                                                                                      10/25/24 18:16:37 PM
    Finished `release` profile [optimized + debuginfo] target(s) in 0.23s
     Running `probe-rs run --chip=esp32c3 target\riscv32imc-unknown-none-elf\release\async_main`
      Erasing ✔ [00:00:00] [#################################################################################] 256.00 KiB/256.00 KiB @ 264.40 KiB/s (eta 0s )
  Programming ✔ [00:00:01] [##################################################################################] 112.23 KiB/112.23 KiB @ 56.61 KiB/s (eta 0s )    Finished in 2.995s
INFO  Hello world!
└─ async_main::____embassy_main_task::{async_fn#0} @ src\bin\async_main.rs:39
INFO  Hello world!
└─ async_main::____embassy_main_task::{async_fn#0} @ src\bin\async_main.rs:39
INFO  Hello world!
└─ async_main::____embassy_main_task::{async_fn#0} @ src\bin\async_main.rs:39
INFO  Hello world!
└─ async_main::____embassy_main_task::{async_fn#0} @ src\bin\async_main.rs:39
INFO  Hello world!
└─ async_main::____embassy_main_task::{async_fn#0} @ src\bin\async_main.rs:39
INFO  Hello world!
└─ async_main::____embassy_main_task::{async_fn#0} @ src\bin\async_main.rs:39
INFO  Hello world!
└─ async_main::____embassy_main_task::{async_fn#0} @ src\bin\async_main.rs:39
INFO  Hello world!
└─ async_main::____embassy_main_task::{async_fn#0} @ src\bin\async_main.rs:39
INFO  Hello world!
└─ async_main::____embassy_main_task::{async_fn#0} @ src\bin\async_main.rs:39
INFO  Hello world!
└─ async_main::____embassy_main_task::{async_fn#0} @ src\bin\async_main.rs:39
INFO  Hello world!
└─ async_main::____embassy_main_task::{async_fn#0} @ src\bin\async_main.rs:39
INFO  Hello world!
└─ async_main::____embassy_main_task::{async_fn#0} @ src\bin\async_main.rs:39
INFO  Hello world!
└─ async_main::____embassy_main_task::{async_fn#0} @ src\bin\async_main.rs:39
INFO  Hello world!
└─ async_main::____embassy_main_task::{async_fn#0} @ src\bin\async_main.rs:39
INFO  Hello world!
└─ async_main::____embassy_main_task::{async_fn#0} @ src\bin\async_main.rs:39
ERROR probe_rs::util::rtt:
Error reading from RTT: The control block has been corrupted. write pointer is 2092342808 while buffer size is 1024 for up channel 0 (defmt)
INFO  Embassy initialized!
└─ async_main::____embassy_main_task::{async_fn#0} @ src\bin\async_main.rs:24
INFO  esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: true, ampdu_tx_enable: true, amsdu_tx_enable: false, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, tick_rate_hz: 100, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 }
└─ esp_wifi::init @ C:\Users\ckren\.cargo\registry\src\index.crates.io-6f17d22bba15001f\esp-wifi-0.10.1\src\fmt.rs:138
 WARN probe_rs::session: Failed to deconfigure device during shutdown: A RISC-V specific error occurred.
```

I will try spawning a separate embassy task and print some logs there.

Greetings, 
Christian

MabezDev, 2024-10-25:
What probe-rs version are you running? I believe we fixed this, at least in git main of probe-rs

Krensi, 2024-10-25:
I will try installing probe-rs with the latest commit of git main

Krensi, 2024-10-25:
Upgrading to `probe-rs 0.24.0 (git commit: feb61572)` fixed it. Thanks for the hint! 👍

closed 2024-10-25 by Krensi