esp-rs/esp-generate #27: Add an argument to choose the template (blocking, async...)

SergioGasquez, 2024-10-29:


bjoernQ, 2025-02-11:
I guess it's worth to discuss this: Is this really something we need and want?

It's taken over from the original cargo-generate template where we haven't had support for async but now we can support multiple `main` files with one template.

We considered this to be used to support additional frameworks like RTIC (or ArielOS or whatever) but I'm not too confident we should even do that - i.e. if we have an option for it in _our_ generator users will expect at least some level of support by us

Maybe there is another use-case I am not thinking of?

SergioGasquez, 2025-02-11:
> I guess it's worth to discuss this: Is this really something we need and want?

Agree, I dont think we need this anymore, current state is good imho. If we change our minds we can always reopen the issue!

closed 2025-02-11 by SergioGasquez