esp-rs/esp-generate #32: Document available options for command-line interface

jessebraham, 2024-10-31:
We allow providing zero or more `--option` arguments when using the CLI, however the available options are not displayed in the help text. We should either find a way to list the options here, or update the description to point users to a list of these options.

bjoernQ, 2024-10-31:
Good point! 👍 I wonder if `clap` has a way to do something like that

(But in general, I would assume users to use the TUI - I mostly thought of the `--option` argument for CI purposes .... but that's no reason to not make it user friendly)

jessebraham, 2024-10-31:
> Good point! 👍 I wonder if `clap` has a way to do something like that

You are able to list available options, however it's a matter of how we accomplish this. Obviously manually maintaining a list is just asking for things to fall out of sync.

> (But in general, I would assume users to use the TUI - I mostly thought of the `--option` argument for CI purposes .... but that's no reason to not make it user friendly)

That's totally reasonable, perhaps this should just be documented somewhere instead, and we can just link to said documentation in the help text. Also we don't necessarily need to move forward with this, I just think there will always be some subset of users which prefer CLI and we should make things a bit easier for them if possible.

PR #36 (2024-11-04): Minor improvements
- list available options in `--help` (closes #32)
- use `target` from `esp-metadata`, avoid duplication
closed 2024-11-04 by bjoernQ