esp-rs/esp-generate #46: Add svd files for debugging

SergioGasquez, 2024-11-14:
See https://github.com/esp-rs/esp-idf-template/issues/96

jessebraham, 2024-11-14:
We had some brief discussion regarding this yesterday. There are some problems here with regards to ensuring that users have the correct version of the SVD.

We are able to tag the PAC releases and use this tool to download the correct version of the SVD from `esp-pacs`, which at least ensures that the correct file is present when generating a project. However, if/when users update their HAL dependency and a new PAC is used, then the SVD will be out of date, and it's not clear how to resolve this.

We will have to spend some more time thinking about how we want to approach this, or if we want to commit to implementing this feature at all.

jessebraham, 2025-03-17:
This should probably just be closed, IMO. There is no clear path forward here based on the multiple discussions we've had previously on this topic.

bjoernQ, 2025-03-17:
I tend to agree - someone who needs them will probably figure out and to most people it's probably not that useful

closed 2025-03-17 by jessebraham