esp-rs/esp-generate #86: Consider setting the MSRV toolchain in rust-toolchain.toml

MabezDev, 2025-01-30:
We could one step further in the rust-toolchain.toml by doing something like this:

```toml
[toolchain]
channel    = "stable-1.84"
components = ["rust-src"]
targets = ["riscv32imc-unknown-none-elf"]
```

Unfortunately I think this will only work on RISC-V until Xtensa is upstreamed.

SergioGasquez, 2025-01-31:
What is the benefit of using the MSRV version instead of the latest stable? It's to have something pinned?

MabezDev, 2025-02-04:
because `stable` doesn't download the latest `stable` toolchain, it just uses the `stable` toolchain installed on the system. I found this when I generated the project and my stable was too old for esp-hal.

bjoernQ, 2025-02-04:
At least #88 would have shown you a red flag ... errr a red cross there.

The one thing I don't like about specific stables or nightlies in `toolchain.toml` is that I end up with a lot of them over time (probably just me especially since my SSDs have enough free space to not worry about it ...)

MabezDev, 2025-02-04:
Yeah to be fair, #88 is better than this solution. Will close this.

closed 2025-02-04 by MabezDev