esp-rs/esp-generate #89: Make sure we reference the right versions of crates, Rust and tools (besides Cargo.toml)

bjoernQ, 2025-02-04:
We have a couple of places where we reference versions

- MSRV in the (not merged yet) tools version check
- That hint about where to look for examples in the generated main
- `espflash` / `probe-rs` versions in the tools version check

Ideally, we can get the MSRV from the `Cargo.toml`

Not sure about `espflash` and `probe-rs` - while users should use a recent version, they might get away with something older that works 🤷‍♂ but we should make sure we suggest a version we know will work

bugadani, 2025-04-02:
I'm somewhat confused what this issue is about exactly, and what the definition of done would be if someone were to tackle this.

bjoernQ, 2025-04-02:
I agree this is probably not an actionable task as is or at least figuring out is part of the task

Could be as simple as having some kind of cfg (in the template.yml ?) where we set those versions instead of having versions set/checked all over the place?

bugadani, 2025-04-02:
Ah so we should store in the template, what the compatible rust compiler, espflash, probe-rs versions are, as opposed to storing them in the tool source? Makes sense, but for probe-rs the CLI API might change a lot still, there are no guarantees there - but perhaps we can signal a "too new" version somehow.

But where do "the right verions of crates" come into play in this issue? Should we try and look up dependencyversions on crates.io? That just sounds like asking for trouble.

bjoernQ, 2025-04-02:
> But where do "the right verions of crates" come into play in this issue? Should we try and look up dependencyversions on crates.io? That just sounds like asking for trouble.

That would probably be a problem - we could do the first part and see if we need to do more (but the first part would close this issue - once we see there is a need to do more we can create a new issue)

bugadani, 2025-08-12:
What else do we have to do here?