esp-rs/esp-generate #105: Add runArgs allowing for flashing from devcontainer

legas7, 2025-02-10:
Hi,

Thanks for this tool! I really like that it also prepares devcontainer config.

Have you considered adding:
```json
  "runArgs": [
    "--privileged",
    "--device=/dev/ttyACM0",
    "--group-add=dialout"
  ],
``` 

As an option in TUI?

SergioGasquez, 2025-02-11:
> As an option in TUI?

I would prefer adding it without an extra option, just include it by default. The thing is, this only works for linux and maybe macos, rigth?

legas7, 2025-02-11:
> The thing is, this only works for linux and maybe macos, rigth?

Seems so. This will add functionality only for linux hosts, but should not break builds for mac/win.

SergioGasquez, 2025-02-12:
> This will add functionality only for linux hosts, but should not break builds for mac/win.

I guess we could add the snippet with some comments if it's not harmful for other OSs, something like:
```json
  # This only works for Linux devices and requires updating the `device` field
  "runArgs": [
    "--privileged",
    "--device=/dev/ttyACM0",
    "--group-add=dialout"
  ],
```

legas7, 2025-02-12:
Checked on win10, running docker desktop. Builds just fine. Don't see why on mac would be different. LGTM!

closed 2025-02-14 by SergioGasquez