esp-rs/esp-generate #121: ESP32C6 template doesn't work

filipw01, 2025-02-13:
I'm a total newbie so maybe I missed some step, but after following steps for [RISC-V Targets Only](https://docs.esp-rs.org/book/installation/riscv.html#risc-v-targets-only) for no_std and generating using esp-generate

The error when building is
```
error[E0583]: file not found for module `arch`
  --> /something/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-backtrace-0.15.0/src/lib.rs:46:1
   |
46 | pub mod arch;
   | ^^^^^^^^^^^^^
   |
   = help: to create the module `arch`, create file "/something/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-backtrace-0.15.0/src/arch.rs" or "/something/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-backtrace-0.15.0/src/arch/mod.rs"
   = note: if there is a `mod arch` elsewhere in the crate already, import it with `use crate::...` instead

error[E0425]: cannot find function `backtrace` in module `crate::arch`
  --> /something/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-backtrace-0.15.0/src/lib.rs:69:34
   |
69 |     let backtrace = crate::arch::backtrace();
   |                                  ^^^^^^^^^ not found in `crate::arch`

error[E0425]: cannot find value `RA_OFFSET` in module `crate::arch`
  --> /something/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-backtrace-0.15.0/src/lib.rs:76:55
   |
76 |         println!("{}0x{:x}", RED, addr - crate::arch::RA_OFFSET);
   |                                                       ^^^^^^^^^ not found in `crate::arch`

Some errors have detailed explanations: E0425, E0583.
For more information about an error, try `rustc --explain E0425`.
error: could not compile `esp-backtrace` (lib) due to 3 previous errors
warning: build failed, waiting for other jobs to finish...
rustc-LLVM ERROR: Global variable '__EXTERNAL_INTERRUPTS' has an invalid section specifier '.rwtext': mach-o section specifier requires a segment and section separated by a comma.
error: could not compile `esp32c6` (lib)

Caused by:
  process didn't exit successfully: `/something/.rustup/toolchains/stable-aarch64-apple-darwin/bin/rustc --crate-name esp32c6 --edition=2021 /something/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32c6-0.18.0/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --diagnostic-width=242 --crate-type lib --emit=dep-info,metadata,link -C opt-level=s -C embed-bitcode=no -C debuginfo=2 -C split-debuginfo=unpacked -C debug-assertions=on --cfg 'feature="critical-section"' --cfg 'feature="default"' --cfg 'feature="rt"' --check-cfg 'cfg(docsrs)' --check-cfg 'cfg(feature, values("critical-section", "default", "defmt", "impl-register-debug", "rt"))' -C metadata=ed02f41b6364d46c -C extra-filename=-ed02f41b6364d46c --out-dir /something/projects/embedded-rust/target/debug/deps -L dependency=/something/projects/embedded-rust/target/debug/deps --extern critical_section=/something/projects/embedded-rust/target/debug/deps/libcritical_section-862735176699bcc5.rmeta --extern vcell=/something/projects/embedded-rust/target/debug/deps/libvcell-96fb4d7b7ba8cf80.rmeta --cap-lints allow -L /something/projects/embedded-rust/target/debug/build/esp32c6-4c51c1fe98eb808a/out` (exit status: 101)
```

It generated

Cargo.toml
```toml
[package]
edition = "2021"
name = "your-project"
version = "0.1.0"

[[bin]]
name = "your-project"
path = "./src/bin/main.rs"

[dependencies]
critical-section = "1.2.0"
esp-backtrace = { version = "0.15.0", features = [
  "esp32c6",
  "exception-handler",
  "panic-handler",
  "println",
] }
esp-hal = { version = "0.23.1", features = ["esp32c6", "unstable"] }
esp-println = { version = "0.13.0", features = ["esp32c6", "log"] }
log = { version = "0.4.21" }

[profile.dev]
# Rust debug is too slow.
# For debug builds always builds with some optimization
opt-level = "s"

[profile.release]
codegen-units = 1        # LLVM can perform better optimizations using a single thread
debug = 2
debug-assertions = false
incremental = false
lto = 'fat'
opt-level = 's'
overflow-checks = false
```

rust-toolchain.toml
```toml
[toolchain]
channel    = "stable"
components = ["rust-src"]
targets = ["riscv32imac-unknown-none-elf"]
```

build.rs
```rust
fn main() {
    println!("cargo:rustc-link-arg=-Tlinkall.x");
}
```

src/bin/main.rs
```rust
#![no_std]
#![no_main]

use esp_backtrace as _;
use esp_hal::clock::CpuClock;
use esp_hal::delay::Delay;
use esp_hal::main;
use log::info;

#[main]
fn main() -> ! {
    // generator version: 0.2.2

    let config = esp_hal::Config::default().with_cpu_clock(CpuClock::max());
    let _peripherals = esp_hal::init(config);

    esp_println::logger::init_logger_from_env();

    let delay = Delay::new();
    loop {
        info!("Hello world!");
        delay.delay_millis(500);
    }

    // for inspiration have a look at the examples at https://github.com/esp-rs/esp-hal/tree/v0.23.1/examples/src/bin
}
```

src/lib.rs
```rust
#![no_std]
```

Any help would be appreciated 😁

SergioGasquez, 2025-02-17:
Could you add the steps you did to install the environment and the esp-generate options that you used to create the project?

filipw01, 2025-02-17:
Oh wow 🤯 That's exactly it. I restored the deleted directory and it has a `.cargo` directory in it. Will try out later today, thank you very much for help