esp-rs/esp-generate #130: CI does not cover all the options

SergioGasquez, 2025-02-20:
The following options are being checked:
- empty, `alloc`, `defmt`, `esp-backtrace`, `log`, `unstable-hal`
We are missing:
- `wifi`, `ble`, `embassy`, `probe-rs`, `flashing-probe-rs/defmt`, `flashing-probe-rs/panic-rtt-target/`, `flashing-espflash/log`, `flashing-espflash/demft`, `flashing-espflash/esp-backtrace`

_Originally posted by @SergioGasquez in https://github.com/esp-rs/esp-generate/pull/129#discussion_r1963645227_

bjoernQ, 2025-02-20:
Given how fast we can do the checks now we can definitely afford to test more

SergioGasquez, 2025-02-20:
The `--all-combinations` is not checking `embassy` at all, neither. So probably we are also missing lots of options there

PR #131 (2025-02-21): Check more
Closes #130
closed 2025-02-21 by SergioGasquez