esp-rs/esp-generate #153: "no external crate `defmt`" on cleanly generated template

VorpalBlade, 2025-04-01:
I get the following build error:

```
error[E0432]: unresolved import `defmt`
  --> /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-io-0.6.1/src/lib.rs:21:41
   |
21 | #[cfg_attr(feature = "defmt-03", derive(defmt::Format))]
   |                                         ^^^^^^^^^^^^^ no external crate `defmt`
   |
   = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0432]: unresolved import `defmt`
  --> /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-io-0.6.1/src/lib.rs:65:41
   |
65 | #[cfg_attr(feature = "defmt-03", derive(defmt::Format))]
   |                                         ^^^^^^^^^^^^^ no external crate `defmt`
   |
   = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0432]: unresolved import `defmt`
   --> /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-io-0.6.1/src/lib.rs:224:41
    |
224 | #[cfg_attr(feature = "defmt-03", derive(defmt::Format))]
    |                                         ^^^^^^^^^^^^^ no external crate `defmt`
    |
    = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0432]: unresolved import `defmt`
   --> /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-io-0.6.1/src/lib.rs:264:41
    |
264 | #[cfg_attr(feature = "defmt-03", derive(defmt::Format))]
    |                                         ^^^^^^^^^^^^^ no external crate `defmt`
    |
    = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0432]: unresolved import `defmt`
   --> /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-io-0.6.1/src/lib.rs:273:41
    |
273 | #[cfg_attr(feature = "defmt-03", derive(defmt::Format))]
    |                                         ^^^^^^^^^^^^^ no external crate `defmt`
    |
    = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0433]: failed to resolve: could not find `defmt` in the list of imported crates
   --> /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-io-0.6.1/src/lib.rs:273:41
    |
273 | #[cfg_attr(feature = "defmt-03", derive(defmt::Format))]
    |                                         ^^^^^^^^^^^^^ could not find `defmt` in the list of imported crates
    |
    = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0433]: failed to resolve: could not find `defmt` in the list of imported crates
   --> /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-io-0.6.1/src/lib.rs:264:41
    |
264 | #[cfg_attr(feature = "defmt-03", derive(defmt::Format))]
    |                                         ^^^^^^^^^^^^^ could not find `defmt` in the list of imported crates
    |
    = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0433]: failed to resolve: could not find `defmt` in the list of imported crates
   --> /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-io-0.6.1/src/lib.rs:224:41
    |
224 | #[cfg_attr(feature = "defmt-03", derive(defmt::Format))]
    |                                         ^^^^^^^^^^^^^ could not find `defmt` in the list of imported crates
    |
    = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0433]: failed to resolve: could not find `defmt` in the list of imported crates
  --> /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-io-0.6.1/src/lib.rs:65:41
   |
65 | #[cfg_attr(feature = "defmt-03", derive(defmt::Format))]
   |                                         ^^^^^^^^^^^^^ could not find `defmt` in the list of imported crates
   |
   = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0433]: failed to resolve: could not find `defmt` in the list of imported crates
  --> /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-io-0.6.1/src/lib.rs:21:41
   |
21 | #[cfg_attr(feature = "defmt-03", derive(defmt::Format))]
   |                                         ^^^^^^^^^^^^^ could not find `defmt` in the list of imported crates
   |
   = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)

Some errors have detailed explanations: E0432, E0433.
For more information about an error, try `rustc --explain E0432`.
error: could not compile `embedded-io` (lib) due to 10 previous errors
warning: build failed, waiting for other jobs to finish...

[similar errors are repeated for a few other crates, like embedded-hal]
```

I selected to generate for an ESP32 (without suffix, since that is what I have lying around) and use espflash + defmt. I also enabled unstable-hal + alloc + wifi + embassy + esp-backtrace + wokwi  + ci + vscode (I *think*, I used the menu system, and it doesn't seem to log my selections anywhere).

```console
❯ esp-generate --version
esp-generate 0.3.1
❯ espup --version
espup 0.14.1
❯ rustc --version
rustc 1.84.0-nightly (b0bd11bcc 2025-01-10) (1.84.0.0)
```

VorpalBlade, 2025-04-01:
Thanks for the quick response! I read that bug and I find myself more confused than before. Reinstalling espup fixed the issue for some people? Or not? If so, is that a workaround I should try?

I have had the esp toolchain installed for a long time but haven't gotten around to actually making a project with it until now (it felt way too unstable the first time I tried). So maybe that is worth a try?

bugadani, 2025-04-01:
I think this should be resolved with defmt 1.0.1. As that is reexported from 0.3.10, a simple `cargo update` should solve your issue.

bugadani, 2025-04-01:
> I think, I used the menu system, and it doesn't seem to log my selections anywhere

As an aside, I've also opened https://github.com/esp-rs/esp-generate/pull/154 so that we will print the applied options to the console. I've been missing this, too.

VorpalBlade, 2025-04-01:
Yes, that seems to have solved it. That seems odd, because cargo-generate didn't crate a Cargo.lock file from what I saw. So the most recent versions should have been picked on first build?

bugadani, 2025-04-01:
The most recent version at the time you opened this issue was probably 1.0.0 :)

bugadani, 2025-04-02:
So since you said it's solved, I'm going to close this. Thanks!

closed 2025-04-02 by bugadani