esp-rs/esp-generate #160: Investigate increased time needed for "check all" (i.e. nightly runs)

bjoernQ, 2025-04-07:
See #158

bugadani, 2025-04-08:
Looking at the S3 build, I think we just have a LOT of configurations and as we'll only have more, the build time will get exponentially worse.

This is the list of builds done for S3 [last night](https://github.com/esp-rs/esp-generate/actions/runs/14324367822/job/40147116660)

```
[2025-04-08T03:22:37Z INFO  xtask] ""
[2025-04-08T03:22:37Z INFO  xtask] "alloc"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embassy, embedded-test, panic-rtt-target, probe-rs, stack-smashing-protection, unstable-hal"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embassy, embedded-test, panic-rtt-target, probe-rs, stack-smashing-protection, unstable-hal, wifi"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embassy, embedded-test, panic-rtt-target, probe-rs, unstable-hal"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embassy, embedded-test, panic-rtt-target, probe-rs, unstable-hal, wifi"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embassy, embedded-test, probe-rs, stack-smashing-protection, unstable-hal"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embassy, embedded-test, probe-rs, stack-smashing-protection, unstable-hal, wifi"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embassy, embedded-test, probe-rs, unstable-hal"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embassy, embedded-test, probe-rs, unstable-hal, wifi"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embassy, esp-backtrace, stack-smashing-protection, unstable-hal"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embassy, esp-backtrace, stack-smashing-protection, unstable-hal, wifi"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embassy, esp-backtrace, unstable-hal"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embassy, esp-backtrace, unstable-hal, wifi"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embassy, panic-rtt-target, probe-rs, stack-smashing-protection, unstable-hal"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embassy, panic-rtt-target, probe-rs, stack-smashing-protection, unstable-hal, wifi"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embassy, panic-rtt-target, probe-rs, unstable-hal"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embassy, panic-rtt-target, probe-rs, unstable-hal, wifi"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embassy, probe-rs, stack-smashing-protection, unstable-hal"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embassy, probe-rs, stack-smashing-protection, unstable-hal, wifi"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embassy, probe-rs, unstable-hal"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embassy, probe-rs, unstable-hal, wifi"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embassy, stack-smashing-protection, unstable-hal"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embassy, stack-smashing-protection, unstable-hal, wifi"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embassy, unstable-hal"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embassy, unstable-hal, wifi"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embedded-test, panic-rtt-target, probe-rs, stack-smashing-protection, unstable-hal"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embedded-test, panic-rtt-target, probe-rs, stack-smashing-protection, unstable-hal, wifi"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embedded-test, panic-rtt-target, probe-rs, unstable-hal"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embedded-test, panic-rtt-target, probe-rs, unstable-hal, wifi"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embedded-test, probe-rs, stack-smashing-protection, unstable-hal"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embedded-test, probe-rs, stack-smashing-protection, unstable-hal, wifi"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embedded-test, probe-rs, unstable-hal"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, embedded-test, probe-rs, unstable-hal, wifi"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, esp-backtrace, stack-smashing-protection, unstable-hal"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, esp-backtrace, stack-smashing-protection, unstable-hal, wifi"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, esp-backtrace, unstable-hal"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, esp-backtrace, unstable-hal, wifi"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, panic-rtt-target, probe-rs, stack-smashing-protection, unstable-hal"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, panic-rtt-target, probe-rs, stack-smashing-protection, unstable-hal, wifi"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, panic-rtt-target, probe-rs, unstable-hal"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, panic-rtt-target, probe-rs, unstable-hal, wifi"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, probe-rs, stack-smashing-protection, unstable-hal"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, probe-rs, stack-smashing-protection, unstable-hal, wifi"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, probe-rs, unstable-hal"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, probe-rs, unstable-hal, wifi"
[2025-04-08T03:22:37Z INFO  xtask] "alloc, ble, defmt, stack-smashing-protection, unstable-hal"
[... cut ...]
[2025-04-08T03:22:37Z INFO  xtask] "panic-rtt-target, probe-rs, unstable-hal"
[2025-04-08T03:22:37Z INFO  xtask] "probe-rs"
[2025-04-08T03:22:37Z INFO  xtask] "probe-rs, stack-smashing-protection"
[2025-04-08T03:22:37Z INFO  xtask] "probe-rs, stack-smashing-protection, unstable-hal"
[2025-04-08T03:22:37Z INFO  xtask] "probe-rs, unstable-hal"
[2025-04-08T03:22:37Z INFO  xtask] "stack-smashing-protection"
[2025-04-08T03:22:37Z INFO  xtask] "stack-smashing-protection, unstable-hal"
[2025-04-08T03:22:37Z INFO  xtask] "unstable-hal"
```

bugadani, 2025-04-08:
Also I'm seeing that we build with optimizations enabled, which is... well, realistic, but sloooow

`Finished `dev` profile [optimized + debuginfo] target(s) in 16.21s`