esp-rs/esp-generate #167: Seperate template from binary

MabezDev, 2025-04-28:
We've discussed this a few times, but I think we need to find a consensus as we'll keep discussing it over and over.

### History

`cargo-generate` uses a separated template system, where the templates are independent from the binary.

### The status quo

In esp-generate, we are bundling the template with the binary. This has some benefits, and some downsides which we'll need to weigh up here.

---

## Bundled

### Advantages

* esp-generate, will _always_ valid project as the template is a snapshot of the ecosystem at a given time
* Template issues are typically easier to debug, just install the users version of esp-generate and input their config
* Project generation works offline

### Disadvantages

* Requires updating esp-generate often
* If a user doesn't update (or misses the fact they have to), they don't get the latest template

## Separated

### Advantages

* One time install (sort of, there can/will be bugs)
* Might allow for custom templates (think 3rd party templates from ecosystem)

### Disadvantages

* With `cargo-generate`, we struggled to easily repro generation issues
  * This is probably solvable by embedding the git hash of the template in the generated project, and having the option to generate from a git hash 
* Requires the internet to generate projects

bugadani, 2025-04-28:
> Requires the internet to generate projects

cargo already needs internet connection to download dependencies. esp-generate uses internet connection to check for updates. The user probably should have the option to refer to a template on their file system, but otherwise do we have some requirement against internet connections?

MabezDev, 2025-04-28:
> cargo already needs internet connection to download dependencies. esp-generate uses internet connection to check for updates. The user probably should have the option to refer to a template on their [...]

No requirements, I was just listing pros/cons as I saw them. I don't think requiring an internet connection is unreasonable at all.

bjoernQ, 2025-04-28:
We probably should provide some way to not rely on an internet connection - I guess pointing to a local template is good enough for that

And a way to choose an older version ofc

MabezDev, 2025-09-29:
Imo, the bundling of the template is working well and separation doesn't offer enough benefits to move away from our current approach.