esp-rs/esp-generate #172: TUI Odd Selection Highlighting

theembeddedrustacean, 2025-05-04:
Using `esp-generate` the highlights for selections are behaving in a strange manner. In the shown screenshot, the red highlight is on a row  different than the selected row in bold "Enable HAL features", which then adds a checkmark on a different row "Enable allocations..." when hitting enter. 

[image]

Managing to check/select the proper row follows an odd pattern, and it's not clear what is actually highlighted.

bjoernQ, 2025-05-05:
Is it in this state initially after launch or do you need to enter some sub menus first?

Could you give some more details
- which version of esp-generate (from crates or some commit from the repository)
- OS (Apparently OSX, which version)
- which terminal
- any relevant OS settings changed?

I have never seen this on Windows at least. I think we have some OSX users but never heard about it so it might be something special to your system

theembeddedrustacean, 2025-05-05:
It operates in a strange state from the start. It also becomes worse when navigating through rows and menus. 

Here are the system details:
- MacOS Sequoia 15.3.1
- zsh shell
- esp-generate version 0.3.1

I'm not sure that I recall changing any OS settings in particular. Are there any other specific OS settings that I should be looking at?

bjoernQ, 2025-05-05:
Apparently the default terminal emulator (Terminal.app) on MacOS can cause some issues: https://github.com/ratatui/ratatui/issues/1387

Don't know what others are using on MacOS (it's been a while since last time I used OSX)

jessebraham, 2025-05-05:
I use Rio as my terminal app and it's working fine. Indeed, the default Terminal.app looks really awful and is basically unusable for me 😅

bugadani, 2025-05-05:
The VSCode terminal also has issues. I'm not quite sure what we can do here, we're just printing characters that get misinterpreted.

theembeddedrustacean, 2025-05-05:
> Apparently the default terminal emulator (Terminal.app) on MacOS can cause some issues: [ratatui/ratatui#1387](https://github.com/ratatui/ratatui/issues/1387)
> 
> Don't know what others are using on MacOS (it's been a while since last time I used OSX)

Yeah, I just switched iTerm2 and it's all good there. In the earlier version, with the same device and settings, I didn't experience the same issues. It seems the emulator doesn't play well with some of the added visuals.

bjoernQ, 2025-05-05:
We could check the env-var `TERM_PROGRAM` and if it's `vscode` or `Apple_Terminal` we could refuse to run in TUI mode (ideally with showing a nice hint to the user)

bugadani, 2025-05-05:
If we know the terminal, we can do better than exiting. I'll draft a PR quick.

jessebraham, 2025-05-05:
That branch still works in my main terminal app (Rio), but is still quite broken in Terminal.app unfortunately. Selection highlighting is behaving very strangely.

bugadani, 2025-05-05:
At least vscode behaves with that PR... thanks for the feedback, I'll try to figure out how to iterate on it

jessebraham, 2025-05-05:
This seems to be much better, however still not quite there. Most of the highlighting weirdness has been resolved, however selected items' text is still illegible. Would be nice to get some more data points here though, I do have quite a bit of shell config/theming so perhaps it's a me-issue.

> If that doesn't help, we might just go with bailing on Apple Terminal.

I personally have no issue with this, pretty much everybody I know who uses macOS uses something else. Maybe be an issue for some people, but we also need to pick our battles here I think.

closed 2025-05-05 by bugadani

SwishSwushPow, 2025-06-06:
Just wanted to chime in to provide one more datapoint: I installed `esp-generate` with #173 and I gotta say that the situation is much improved compared to before. I don't run any custom styling, just the default zsh terminal on macOS 15.5 as far as I'm aware. I greatly appreciate the effort made here and wanted to stress that supporting the default terminal is, in my eyes, an important topic to not lose interested users during the first steps of diving into embedded programming. 😅 But I think with this fix a good enough state has been reached already. 👌 

[image]

PR #173 (2025-05-05): Apply fallback style for quirky terminals
Fixes esp-generate in the vscode terminal. Applies a different color scheme for apple terminal
Closes #172