esp-rs/esp-generate #175: Checking installed version: "❌ Rust"

vikblom, 2025-05-21:
Hi,

I'm using this tool to generate a config with just defaults. Specifcally 
```
> esp-generate --version
esp-generate 0.3.1
```

After creation the tool prints:
```
> esp-generate --chip=esp32s3 -O . foo 

Checking installed versions
❌ Rust
🆗 espflash
🆗 probe-rs
```

What is check is behind the "❌ Rust" and how can I fix that?

It would be helpful if this information was included in the printout. 

Thanks!

bugadani, 2025-05-21:
Please give us the output of `rustc +esp --version`

bjoernQ, 2025-05-22:
The check is here: https://github.com/esp-rs/esp-generate/blob/e0cfa7b54ab8fd30b182f54f8ea1d922ce991098/src/check.rs#L19-L45

The `X` means not found at all

Did you source `export-esp-sh` before running `esp-generate`?

bugadani, 2025-05-22:
Let's keep this open until we add some advice what the various things mean.

PR #178 (2025-05-27): Explain version check results
Closes #175

I was thinking about giving more details (the required version and/or the actual command executed) but maybe that is too much noise and not too helpful

Anyways just relying on emojis was a bit too brief