esp-rs/esp-generate #194: Bootloop after building from esp-generate template using embassy + WiFi/BLE + alloc

Axel-ex, 2025-06-12:
### Description:

I’ve encountered a bootloop after building a project using the esp-generate template with Embassy support. The board continuously reboots with the following error message:

```
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7104
0x3fff0030 - test_re::____embassy_main_task::{{closure}}::HEAP
    at ??:??
load:0x40078000,len:15576
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3876
entry 0x4008064c
0x4008064c - esp_hal_embassy::executor::thread::Executor::wait_impl
    at /Users/Axel/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-hal-embassy-0.8.1/src/executor/thread.rs:136
I (60) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (60) boot: compile time Jun  7 2023 07:48:23
I (63) boot: Multicore bootloader
I (67) boot: chip revision: v3.1
I (71) boot.esp32: SPI Speed      : 40MHz
I (76) boot.esp32: SPI Mode       : DIO
I (80) boot.esp32: SPI Flash Size : 4MB
I (85) boot: Enabling RNG early entropy source...
I (90) boot: Partition Table:
I (94) boot: ## Label            Usage          Type ST Offset   Length
I (101) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (109) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (116) boot:  2 factory          factory app      00 00 00010000 003f0000
I (124) boot: End of partition table
I (128) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=16288h ( 90760) map
I (169) esp_image: segment 1: paddr=000262b0 vaddr=3ffc0000 size=0255ch (  9564) load
I (173) esp_image: segment 2: paddr=00028814 vaddr=3ffdb52c size=00734h (  1844) load
E (175) esp_image: Segment 2 0x3ffdb52c-0x3ffdbc60 invalid: overlaps bootloader stack
E (184) boot: Factory app partition is not bootable
E (189) boot: No bootable app partitions in the partition table

```

xtensa-esp32s2-elf-size and xtensa-esp32s2-elfobjdump output:

```
 xtensa-esp32s2-elf-size -Ax target/xtensa-esp32-none-elf/debug/test_re
target/xtensa-esp32-none-elf/debug/test_re  :
section                    size         addr
.rodata_desc              0x100   0x3f400020
.rwtext                  0x1988   0x40080400
.rwtext.wifi            0x11d7c   0x40081d88
.data                    0x255c   0x3ffc0000
.bss                    0x18fcc   0x3ffc2560
.noinit                     0x0   0x3ffdb52c
.data.wifi                0x734   0x3ffdb52c
.rodata                  0xfa28   0x3f400120
.rodata.wifi             0x6760   0x3f40fb48
.text                   0x5ee19   0x400d0020
.coexsleepiram.26           0xe   0x4012ee3c
.wifiorslpiram.27          0x2c   0x4012ee4c
.rtc_fast.text              0x0   0x400c0000
.vectors                  0x400   0x40080000
.rtc_fast.dummy             0x0   0x3ff80000
.rtc_fast.data              0x0   0x3ff80000
.rtc_fast.bss               0x0   0x3ff80000
.rtc_fast.persistent        0x0   0x3ff80000
.rtc_slow.text              0x0   0x50000000
.rtc_slow.data              0x0   0x50000000
.rtc_slow.bss               0x0   0x50000000
.rtc_slow.persistent        0x0   0x50000000
.stack                    0x3a0   0x3ffdbc60
.dram2_uninit           0x10000   0x3ffe7e30
.espressif.metadata        0x12          0x0
.debug_loc              0xe203b          0x0
.debug_abbrev           0x205f3          0x0
.debug_info            0x2b02e6          0x0
.debug_aranges           0x96c8          0x0
.debug_ranges           0x3a698          0x0
.debug_str             0x19d063          0x0
.comment                  0x11c          0x0
.xtensa.info               0x38          0x0
.debug_line             0xb8c52          0x0
.debug_frame             0x4cb0          0x0
Total                  0x705d1a

```
```
xtensa-esp32s2-elf-objdump -h target/xtensa-esp32-none-elf/debug/test_re | grep -i wifi

  2 .rwtext.wifi  00011d7c  40081d88  40081d88  00035d88  2**2
  6 .data.wifi    00000734  3ffdb52c  3ffdb52c  0003352c  2**2
  8 .rodata.wifi  00006760  3f40fb48  3f40fb48  00010b48  2**2
 11 .wifiorslpiram.27 0000002c  4012ee4c  4012ee4c  000a6e4c  2**2

```

### Observation:

- It seems like .data.wifi overlaps the bootloader area.
- This problem arises without spawning any task.
- The problem seems to arise with the use of esp-wifi 0.14.1

### To Reproduce:

Generate a project with esp-generate for esp32 chip enabling: 
- Unstable HAL features
- alloc via esp-alloc
- Wifi
- BLE (both options lead to the same issue)
- embassy framework support
- espflash and log crate.

The chip reboots continuously with the above error about overlapping with the bootloader stack.

### setup
rustc 1.86.0-nightly (5944fc5cd 2025-04-01) (1.86.0.0)
esp-generate 0.4.0
xtensa-esp32s2-elf-size  2.43.1
xtensa-esp32s2-elf-objdump 2.43.1

bugadani, 2025-06-12:
Can you please also post your espflash version?

Axel-ex, 2025-06-12:
Sure i am using espflash 3.3.0!

bugadani, 2025-06-12:
The issue will be fixed in esp-hal (https://github.com/esp-rs/esp-hal/pull/3635). The cause is improper memory layout ending up placing data where it shouldn't. A better workaround than `--release` is changing how you allocate heap, to prevent accidental stack overflows/memory corruptions:

```rust
esp_alloc::heap_allocator!(size: 32 * 1024); // Place less in the default region, to ensure you have adequate stack space
esp_alloc::heap_allocator!(#[unsafe(link_section = ".dram2_uninit")] size: 96 * 1024); // Allocate more in the reclaimed region to keep the total heap around the same size.
```

Axel-ex, 2025-06-13:
Amzing, thank you for the insights and you reactivity. Cheers!

closed 2025-06-13 by Axel-ex